rapyuta-robotics/rclUE #30: Generators for the Msg C++ classes

russkel, 2022-02-01:
Hello,

Thanks for publishing your work.

Are you please able to share the packages used to auto-generate the message C++ files, such as https://github.com/rapyuta-robotics/rclUE/blob/master/Source/rclUE/Private/Msgs/ROS2Int64Msg.cpp

Thanks,

Russ

russkel, 2022-02-03:
And is the `.msg` files available for the `ue_msgs` package?

russkel, 2022-02-10:
Ping @yuokamoto

yuokamoto, 2022-02-15:
Sorry for late. please check follwing repos.
*those repos are experimental. let us know if you see issues.

msg definitions: https://github.com/rapyuta-robotics/UE_msgs
auto code geneation script: https://github.com/rapyuta-robotics/UE_tools

russkel, 2022-02-15:
Thank you for this @yuokamoto!

I have started using this plugin internally for a project and have quite a lot of changes (including using external ROS2 install).

closed 2022-05-16 by yuokamoto