rapyuta-robotics/rclUE #37: Timestamp is not accurate

james-yoo, 2022-03-26:
Hi rapyuta-robotics team,

I am testing rclUE plugin with a example video clip you provided. I have set topic(clock msg) publish rate to 100Hz but I can not get correct timestamp. So I checked a video clip and it also showed inaccurate timestamp as below:
[image]

I would like to know how to deal with this issue because accurate timestamp is very important in my application.

james-yoo, 2022-03-31:
ping @yuokamoto

chalice-graeme, 2022-04-01:
The publisher uses an Unreal timer for the update. These are not asynchronous, but they attempt to match the update frequency you supply. If your framerate is lower than the update frequency you won't get wall-clock accurate timing. Some things to be aware of: Unreal uses a variable timestep by default. You can change it to a fixed tick in your project settings. If your machine cannot do 100fps, then it will drop frames to keep the update rate satisfied. This might still cause inaccuracies in the timer callback. Also, the editor behaves on its own and doesn't honor project settings for update rate. If I recall, it's 60hz. It also drops to 15hz if it's not in focus. If you want to see something at 100hz you will need to build and run the standalone game and check that.

james-yoo, 2022-04-01:
@chalice-graeme 
Thanks you for your detailed explanation. I'll try change settings and game mode. 
Thanks.

closed 2022-04-01 by james-yoo