rapyuta-robotics/rclUE #49: Service and Action example

james-yoo, 2022-06-15:
@yuokamoto,

There is only pubsub example in the documentation(https://rclue.readthedocs.io/en/devel/examples.html).
I would like to know how to apply service and action interface using rclUE.

yuokamoto, 2022-06-17:
Got it. we will work on it. In the mean time
- you can find ROS2 service C++ example in SimulationState class. https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/4eaf79c39b12d6fe15a986214b974b537c8ab45b/Source/RapyutaSimulationPlugins/Public/Tools/SimulationState.h 
- Action has bug and need to be fixed.

james-yoo, 2022-10-13:
@yuokamoto 
Thanks for your active work. Any update on documentation?

yuokamoto, 2022-10-22:
Sorry for late. We were busy for ROSCon but will restart updating documentation.

james-yoo, 2023-04-01:
Issue close by commit https://github.com/rapyuta-robotics/rclUE/commit/a37e905a1fdc59e7a712e47234f859f69b5bbc20

closed 2023-04-01 by james-yoo