rapyuta-robotics/rclUE #61: Using ROS2 Humble

Remi-Niel, 2022-10-20:
Hi Rapyuta-robotics team,

I was wondering whether ROS2 Humble is or will be supported since it si next LTS version of ROS2 and Foxy is approaching EOL date.

yuokamoto, 2022-10-22:
Hello, Thanks to have interested in our project.
Yes, we are planning to support ROS2 Humble. Since we were busy for ROSCon, we will work on Humble support later.

tomkimsour, 2022-11-29:
I'd be very interested in a humble version as well. If that ever comes

tomkimsour, 2022-12-05:
If anyone is interested. I am trying to make it work and compile with ros humble on unreal 5.1

yuokamoto, 2022-12-05:
We are working on that too. Currently following repos have UE5_devel branch we are testing with UE5.1
- turtlebot3-UE 
- RapyutaSimulationPluginsh

rclUE should work with devel branch but we are not tested many time/many cases yet.
cc @Tadinu

yuokamoto, 2022-12-07:
I created rclUE branch UE5_devel_humble. Havne't tested fully but build pass and ROS2Examples levels works.

yuokamoto, 2023-04-29:
This cussion is keep happening in
https://github.com/rapyuta-robotics/rclUE/issues/78

closed 2023-04-29 by yuokamoto