rapyuta-robotics/rclUE #74: Warning: dlopen failed: libpython3.8.so.1.0: cannot open shared object file: No such file or directory

tomkimsour, 2022-12-15:
after running ```./run_editor.sh /home/crossing/Documents/Unreal\ Projects/turtlebot3-UE\ 5.0 /home/crossing/UnrealEngine/Engine/Binaries/Linux/UnrealEditor```

I am getting this error :
```
LogPixelStreaming: Initialising Pixel Streaming settings.
LogAudio: Display: Registering Engine Module Parameter Interfaces...
LogMetasoundEngine: MetaSound Engine Initialized
LogPython: Using Python 3.9.7
LogCore: Warning: dlopen failed: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

However the only version of python installed on my computer is 3.10.6 , is there a way to upgrade the dependancy of the file ? If so, which file are we looking at and how is it even able to find python 3.9.7 ?

yuokamoto, 2022-12-16:
- @chalice-graeme do you have any idea?
- @tomkimsour does pyenv help you to use different python version?

chalice-graeme, 2022-12-16:
Unreal embeds Python. That is likely what you're seeing with regards to the version, but I have no idea why it wouldn't be able to find a library for it.

james-yoo, 2023-01-01:
I had exactly same issue and resolved by install 'python3.8' and 'python3.8-dev'.
However, another issue has occurred regarding of 'RMW' as below:

[ERROR] [1672537372.036080733] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library 'librmw_fastrtps_cpp.so' due to dlopen error: libtinyxml2.so.6: cannot open shared object file: No such file or directory, at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rcutils/src/shared_library.c:99, at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:67', exiting with 1., at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rcl/rcl/src/rcl/rmw_implementation_identifier_check.c:145  

I am testing under following condition:
- Ubuntu 22.04
- ROS2 Humble (binary install)
- Unreal Engine 5.1.0
- turtlebot3-UE(branch: devel)
- rclUE(branch: UE5_devel_humble)
- RapyutaSimulationPlugins (branch: devel)

closed 2023-01-05 by tomkimsour