rapyuta-robotics/rclUE #78: RTPS errors while running humble

tomkimsour, 2023-01-05:
A similar error has been mentioned in #74 

I'm getting similar error regarding fastrtps now as well. The project starts but no topic is published.
After running the running command I get 
```
2023-01-05 15:23:58.290 [RTPS_PARTICIPANT Error] Discovery Server wasn't able to allocate the specified listening port. -> Function createParticipant
2023-01-05 15:23:58.291 [DOMAIN_PARTICIPANT Error] Problem creating RTPSParticipant -> Function enable
```

the command ```fastdds discovery -i 0``` is returning this error. I think this is what leads to no topic being published once unreal started.

tomkimsour, 2023-01-09:
Alright this problem is just the ros daemon already running. If you run ```ros2 daemon stop``` the error will be removed on the next run.

However, the following error is still there 
```
Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library 'librmw_fastrtps_cpp.so' due to dlopen error: libtinyxml2.so.6: cannot open shared object file: No such file or directory, at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rcutils/src/shared_library.c:99, at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:67', exiting with 1., at /home/yu/UE_tools/BuildROS2/ros2_ws/src/ros2/rcl/rcl/src/rcl/rmw_implementation_identifier_check.c:145
```

@james-yoo I think I understand what is happening. The .so files generated use the path of compilation of the machine. So if you haven't generated rclUE with your own .so files, unreal will look under the path /home/yu which is likely not a path existing on your computer. Therefore your need to generate the .so files using UE_tools .

yuokamoto, 2023-01-09:
Thanks for reporting.
hmm I'm not sure since our team member can use simulation without building ros2 lib by themselves. I'll look into the issue later

james-yoo, 2023-01-09:
@tomkimsour Thanks for point me out. However, this is weird as @yuokamoto said.

tomkimsour, 2023-01-15:
@yuokamoto were you able to find a way to solve this ?
Also, I've been trying to improve how the dockerfile compiles files but somehow it is not compiling anymore so I've been on a stalemate for some time.

yuokamoto, 2023-01-16:
I can't reproduce issue and not able to solve yet.

I just started testing ros2 build inside docker. 
https://github.com/rapyuta-robotics/UE_tools/commit/d4b5bee25a7c2d35e61272cdee8212888ae870c5

yuokamoto, 2023-02-05:
I created pr for UE_tools and rclUE
https://github.com/rapyuta-robotics/UE_tools/pull/17
https://github.com/rapyuta-robotics/rclUE/pull/81

Since I don't have ubuntu 22.04 machine here, I can't test but I confirmed rclUE devel branch work with foxy and UE5_devel_humble_20.04 branch works with humble with Ubuntu 20.04.

james-yoo, 2023-02-05:
@yuokamoto 
This PR works under Ubuntu 22.04 with UE 5.1.
[image]

tomkimsour, 2023-02-06:
I will give it a try if I find time. Thanks for the PR tho

james-yoo, 2023-02-14:
@yuokamoto 
OK, now the map is correctly loaded and transforms are set as follows.
[image]
However, after setting up the initial pose of the robot, the following error message is showing in the terminal.
```
     :
[rviz2-3] [ERROR] [1676418840.850748802] [rviz2]: Lookup would require extrapolation into the future.  Requested time 66.853240 but the latest data is at time 66.849998, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [1676418840.913146018] [rviz2]: Lookup would require extrapolation into the future.  Requested time 66.883239 but the latest data is at time 66.869995, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [1676418840.945514027] [rviz2]: Lookup would require extrapolation into the future.  Requested time 66.913238 but the latest data is at time 66.889999, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [1676418841.009329278] [rviz2]: Lookup would require extrapolation into the future.  Requested time 66.953239 but the latest data is at time 66.930000, when looking up transform from frame [base_scan] to frame [map]
     :
```

hirasawaryo, 2023-04-14:
Hello. 
I'm getting the same error of the top of this conversation. I can start the unreal project, but no topic being published. In my case, `ros2 daemon stop` cannot free us from this problem. Help me, please.

```
2023-04-15 05:53:22.169 [RTPS_PARTICIPANT Error] Discovery Server wasn't able to allocate the specified listening port. -> Function createParticipant
Server creation failed with the given settings. Please review locators setup.
2023-04-15 05:53:22.171 [DOMAIN_PARTICIPANT Error] Problem creating RTPSParticipant -> Function enable
The daemon is already running
fastdds discovery <discovery-args>

1
 fast-discovery-server tool not found!
```

yuokamoto, 2023-04-15:
@hirasawaryo Hello, thanks for reporting.
- Are you using Ubunto 20.04 or 22.04?
- if you are using Ubuntu 22.04, could you try this branch https://github.com/rapyuta-robotics/turtlebot3-UE/tree/option_dds_server and start project with ./run_editor.sh false in new terminal? this will start project without dds server

hirasawaryo, 2023-04-17:
Thanks for your reply.
I use Ubuntu 22.04. I cannot reach the URL you teach me. Is this branch's name correct?

yuokamoto, 2023-04-17:
Sorry, branch was merged. Please try `jammy` branch in turtlebot3-UE
https://github.com/rapyuta-robotics/turtlebot3-UE/tree/jammy

yuokamoto, 2023-08-12:
closed with no activity

closed 2023-08-12 by yuokamoto