rapyuta-robotics/rclUE #83: error - rcl node implementation is invalid

james-yoo, 2023-02-25:
Hi @yuokamoto,
I am currently testing under following condition:
- Ubuntu 22.04
- UE 5.1
- turtlebot3-UE(branch: devel)
- rclUE(branch: UE5_humble_devel_22.04)
- RapyutaSimulationPlugins(branch: devel)

When I start simulation(level: turtlebot3_benchmark), I can see following topics:
```
$ ros2 topic list
/clock
/cmd_vel
/joint_states
/odom
/parameter_events
/rosout
/scan
/tf
```
However, I cannot receive the topic data and I found the following error on the terminal.
```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'rcl node implementation is invalid, at /home/admin/UE_tools/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:414'

with this new error message:

  'Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at /home/admin/UE_tools/ros2_ws/src/ros2/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:113'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
```
Any suggestion on resolve this issue?

yuokamoto, 2023-02-25:
Didn't you able to run sim before as this comment?
https://github.com/rapyuta-robotics/rclUE/issues/78#issuecomment-1430549551

james-yoo, 2023-02-25:
@yuokamoto 
Recently, I reinstalled ubuntu 22.04 and ros2 humble from binary. I also tried rclUE with UE5_humble_devel branch,  it failed compile.

james-yoo, 2023-02-27:
To check if it was the latest version issue of the ros2 humble, I tested it with the following two versions:
- humble patch2(https://github.com/ros2/ros2/releases/tag/release-humble-20230127)
- humble patch1(https://github.com/ros2/ros2/releases/tag/release-humble-20221123)

Unfortunately, the results were the same as follows:
```
[2023.02.27-04.19.13:568][431]LogAudio: Display: Audio Device (ID: 2) registered with world 'Turtlebot3_benchmark'.
[2023.02.27-04.19.13:570][431]LogLoad: Game class is 'RRROS2GameMode'
[2023.02.27-04.19.13:571][431]LogWorld: Bringing World /Game/Maps/UEDPIE_0_Turtlebot3_benchmark.Turtlebot3_benchmark up for play (max tick rate 0) at 2023.02.27-13.19.13
[2023.02.27-04.19.13:571][431]LogRapyutaCore: [ARRROS2GameMode] INIT GAME [UEDPIE_0_Turtlebot3_benchmark/Turtlebot3_benchmark] - Options: 

[2023.02.27-04.19.13:571][431]LogRapyutaCore: NUM OF CPU CORES: [8] - WITH HYPERTHREADS: [16] - RECOMMENDED NUM OF WORKER THREADS: [14]
[2023.02.27-04.19.13:571][431]LogRapyutaCore: Display: ShouldUseThreadingForPerformance: 1
[2023.02.27-04.19.13:572][431]LogROS2Node: [ROS2NodeComponent] start initializing.. (ROS2NodeComponent.cpp::Init::107)
[2023.02.27-04.19.13:572][431]LogROS2Node: [ROS2NodeComponent] rclc_node_init_default (ROS2NodeComponent.cpp::Init::112)

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'rcl node implementation is invalid, at /home/admin/UE_tools/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:414'

with this new error message:

  'Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at /home/admin/UE_tools/ros2_ws/src/ros2/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:113'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[2023.02.27-04.19.13:583][431]LogROS2Node: [ROS2NodeComponent] ROS2Node inited (ROS2NodeComponent.cpp::Init::118)
[2023.02.27-04.19.13:590][431]LogRapyutaCore: Display: [BP_TurtlebotBurgerVehicle_C_0][ARRBaseRobot::CreateROS2Interface] IsNetMode: 0
[2023.02.27-04.19.13:590][431]LogRapyutaCore: Warning: [URRGameSingleton] RF_Standalone!!
[2023.02.27-04.19.13:590][431]LogRapyutaCore: Warning: [BP_TurtlebotBurgerVehicle_C_0ROS2Interface] [URRRobotROS2Interface::Initialize] InitializeROS2Node
[2023.02.27-04.19.13:590][431]LogROS2Node: [ROS2NodeComponent_0] start initializing.. (ROS2NodeComponent.cpp::Init::107)
[2023.02.27-04.19.13:590][431]LogROS2Node: [ROS2NodeComponent_0] rclc_node_init_default (ROS2NodeComponent.cpp::Init::112)
[2023.02.27-04.19.13:590][431]LogROS2Node: [ROS2NodeComponent_0] ROS2Node inited (ROS2NodeComponent.cpp::Init::118)
[2023.02.27-04.19.13:595][431]LogTemp: [URRTimerManager::SetTimer][scan] Timer stopped
[2023.02.27-04.19.13:595][431]LogTemp: [URRTimerManager::SetTimer][scan] Timer started
[2023.02.27-04.19.13:599][431]LogTemp: [URRTimerManager::SetTimer][odom] Timer stopped
[2023.02.27-04.19.13:599][431]LogTemp: [URRTimerManager::SetTimer][odom] Timer started
[2023.02.27-04.19.13:604][431]LogTemp: [URRTimerManager::SetTimer][/tf] Timer stopped
[2023.02.27-04.19.13:604][431]LogTemp: [URRTimerManager::SetTimer][/tf] Timer started
[2023.02.27-04.19.13:607][431]LogRapyutaCore: Display: [BP_TurtlebotBurgerVehicle_C_0MoveComp] created from class RobotVehicleMovementComponent!
[2023.02.27-04.19.13:607][431]LogWorld: Bringing up level for play took: 0.036634
[2023.02.27-04.19.13:608][431]LogOnline: OSS: Created online subsystem instance for: :Context_1
[2023.02.27-04.19.13:621][431]LogGameMode: Display: Match State Changed from EnteringMap to WaitingToStart
[2023.02.27-04.19.13:621][431]LogGameState: Match State Changed from EnteringMap to WaitingToStart
[2023.02.27-04.19.13:621][431]LogGameMode: Display: Match State Changed from WaitingToStart to InProgress
[2023.02.27-04.19.13:622][431]LogViewport: Display: Viewport MouseLockMode Changed, LockOnCapture -> DoNotLock
[2023.02.27-04.19.13:622][431]LogViewport: Display: Viewport MouseCaptureMode Changed, CapturePermanently_IncludingInitialMouseDown -> CaptureDuringMouseDown
[2023.02.27-04.19.13:622][431]LogGameState: Match State Changed from WaitingToStart to InProgress
[2023.02.27-04.19.13:622][431]LogRapyutaCore: Display: [ARRROS2GameMode]: START PLAY!
[2023.02.27-04.19.13:622][431]PIE: Server logged in
[2023.02.27-04.19.13:622][431]PIE: Play in editor total start time 0.199 seconds.
[2023.02.27-04.19.45:863][634]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2023.02.27-04.19.45:863][634]LogWorld: BeginTearingDown for /Game/Maps/UEDPIE_0_Turtlebot3_benchmark
[2023.02.27-04.19.45:863][634]LogROS2Topic: Publisher Destroy - rcl_publisher_fini  (ROS2Publisher.cpp::Destroy::108)
[2023.02.27-04.19.45:867][634]LogROS2Srv: Service server Destroy - rcl_service_fini  (ROS2ServiceServer.cpp::Destroy::36)
[2023.02.27-04.19.45:869][634]LogROS2Srv: Service server Destroy - rcl_service_fini  (ROS2ServiceServer.cpp::Destroy::36)
[2023.02.27-04.19.45:870][634]LogROS2Srv: Service server Destroy - rcl_service_fini  (ROS2ServiceServer.cpp::Destroy::36)
[2023.02.27-04.19.45:870][634]LogROS2Srv: Service server Destroy - rcl_service_fini  (ROS2ServiceServer.cpp::Destroy::36)
[2023.02.27-04.19.45:871][634]LogROS2Srv: Service server Destroy - rcl_service_fini  (ROS2ServiceServer.cpp::Destroy::36)
[... cut ...]
[2023.02.27-04.19.45:875][634]LogPlayLevel: Display: Shutting down PIE online subsystems
[2023.02.27-04.19.45:877][634]LogSlate: InvalidateAllWidgets triggered.  All widgets were invalidated
[2023.02.27-04.19.45:883][634]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2023.02.27-04.19.45:918][634]LogUObjectHash: Compacting FUObjectHashTables data took   0.42ms
[2023.02.27-04.19.46:026][645]LogPlayLevel: Display: Destroying online subsystem :Context_1
```

james-yoo, 2023-02-28:
@yuokamoto 
I used UE_Tools to generate the library on my local PC, but I still get the same error. I'm looking for the reason why it worked last  
 time #78, but It is hard to figure it out.

james-yoo, 2023-03-02:
@yuokamoto 
Any updates?

yuokamoto, 2023-03-02:
Sorry for not responding.
I may have time weekend or next week to investigate more. but at least I havn't been able to reproduced yet.

james-yoo, 2023-03-02:
@yuokamoto 
Thanks for your response. 
Did you install ros2 humble from binary or compile from source? I also tried with 'UE5_devel_humble_20.04' branch on ubuntu 20.04 machine, but same error occurred.

yuokamoto, 2023-03-02:
- 22.04, I install humble from binary, huble is `Unpacking ros-humble-desktop (0.10.0-1jammy.20230221.193332) over (0.10.0-1jammy
.20230124.081312)`
- 20.04 I used Docker since I failed to build humble in 20.04
- In my new computer, I didn't build lib with UE-tools but just pull the rclUE repo
- I pushed UE5_devel_humble_22.04 to sync with changes in devel branch but this shouldn't affect since it is just documentation/log update

yuokamoto, 2023-03-02:
I noteiced there is same error as you but navigation is somehow working.
Could you try comment out line 13 and 16 in the run_editor.sh as below?

```
# # Run discovery service for FastDDS
# (exec "${TB3_UE_DIR}/run_discovery_service.sh")

# # Configure environment for FastDDS discovery
# source ${TB3_UE_DIR}/fastdds_setup.sh

james-yoo, 2023-03-02:
@yuokamoto 
Thanks for your advice. I confirmed that it work by commenting out that lines.

yuokamoto, 2023-03-03:
Sorry for inconvinience. those lines are introduced to make ros2 communication with packaged simulation inside docker. 
We will test more on humble and maybe removed if not necessary.

closed 2023-03-03 by yuokamoto