rapyuta-robotics/rclUE #94: Windows 10

hirasawaryo, 2023-06-06:
Do you have a plan to support windows 10? Unreal Engine 5 has many useful functions (Nanite, Lumen, etc) which are supported only by Windows OS. Currently ROS2 supports windows 10, not windows 11. I hope you to support windows 10.

yuokamoto, 2023-06-07:
We want to support windows as well but currently do not have resource to work on that.  It requires much more understanding on ROS2 build on windows....

hirasawaryo, 2023-06-08:
I see. Please continue to consider supporting windows 10.

NicoBrug, 2023-10-20:
Hi,

I have ported the plugin to Windows. You have to get all the ROS packages in lib/dll  and add a specific dll loading process. 

Some libraries are not accessible from the ROS binary installation. You have to download the ROS sources and do a manual installation to get the missing packages (really pain in the ass on windows).

yuokamoto, 2023-10-21:
Thank you!
Could you share the process/branch if possible? then we can work on adding process to script in this repo https://github.com/rapyuta-robotics/UE_tools 
and create branch for Windows.
cc @chalice-graeme

NicoBrug, 2023-11-06:
Hi, 
You need to install the ROS dependencies and install it on your Windows system. For this, I recommend the documentation on Humble, which is quite clear. (https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html and https://github.com/ros2/ros2/releases). I have only tested with the windows debug version.

So you install ROS2 Humble from the binaries without recompiling. However, on Windows, the ROS binaries do not contain RCLC. For that you have to get it on github (https://github.com/ros2/rclc) and recompile it by hand (with cmake command). There is a dependency with rcutils to remember.

You now have all the files you need: take the /include /Bin /Lib files corresponding to your libraries (.so) and put them in the project.

I didn't make a process to do all this automatically, I did it all by hand (but it's not too complicated to script). The compilation process is fairly simple if you have all the environment installed (visual studio 2022, ros dependencies etc).

To load them onto Unreal, the method is pretty much the same as with Linux, you need to :

Include the .h files: 

```
PublicIncludePaths.Add(include);
```

Add the libs 
```
              foreach (var lib in libsPathList)
                {
                    System.Console.WriteLine(lib);
                    PublicAdditionalLibraries.Add(lib);
                }
```
Adding dlls 

```
                foreach (var dll in dllsPathList)
                {
                    System.Console.WriteLine(dll);
                    PublicDelayLoadDLLs.Add(dll);
                }
```

Use RuntimeDependencies if you want them in the package. 

Importantly, I didn't test on windows in depth, I just checked the compilation with windows dlls and libs and simply made sure I didn't have any runtime crashes. If you encounter any problems, please contact me. I don't claim to be an expert, but I might have encountered the problem.

james-yoo, 2023-11-07:
Plus, ue_msgs package (https://github.com/rapyuta-robotics/UE_msgs) should be compiled.

phs4530, 2023-11-07:
When building in Windows10, an error occurred due to a uint size issue.
If you want to build on Windows, cherry-pick the commits for the relevant modified code.
- https://github.com/rapyuta-robotics/rclUE/pull/107

soul667, 2024-01-19:
i think you can try this to install ros2 in windows ,all you need to do is type choco source add -n=ros-win -s="https://aka.ms/ros/public" --priority=1,and wait. and i use humble,so amazing.

tryan-phantom, 2024-03-15:
> Unreal Engine 5 has many useful functions (Nanite, Lumen, etc) which are supported only by Windows OS.

For anyone coming across this, Nanite and Lumen are supported on Linux as well.  I am unaware if there are limitations, but I have tested using UE 5.3.2 on Ubuntu 22.04 via Docker.

james-yoo, 2024-03-16:
> 
> For anyone coming across this, Nanite and Lumen are supported on Linux as well. I am unaware if there are limitations, but I have tested using UE 5.3.2 on Ubuntu 22.04 via Docker.
There are powerful plugins only available on Windows that boost rendering performance such as DLSS 3.5 or FSR 3.

PeteBaker1, 2024-04-16:
Hi @NicoBrug, do you have a github repo with what the bare bones unreal engine project should look like once these steps are completed?

arlottandrea, 2025-12-04:
Hello everybody! have someone tried to do a port Windows while using ROS2 Jazzy? any suggestion would be appreciated 🙏