rapyuta-robotics/rclUE #95: What steps should I take if I want to run turtlebot3 in ROS2 using an environment I created myself in unreal engine?

YuminosukeSato, 2023-06-10:
I created my own environment with unreal engine. And there I would like to build a route for turtlebot3 using navigation in ROS2. So here is my question.
What steps should I take if I want to run turtlebot3 in ROS2 using an environment I created myself in unreal engine?

yuokamoto, 2023-06-10:
I haven't tried but theoritically
1. add RapyutaSimulationPlugins and rclUE as plugin
2. use ROS2GameMode which publish /clock topic and provide ROS 2 services for control Actors
3. place turtlebot3 from editor or spawn from /SpawnEntity ROS2 service.

YuminosukeSato, 2023-06-10:
There is a repository called turtlebot3-UE, but is it possible to change the environment other than turtlebot (for example, an environment like turtlebot3_benchmark) to another building or space?

yuokamoto, 2023-06-10:
Yes. you can fork the repository and edit as you like. 
turtlebot3_benchmark has [OccupancyMapGenerator](https://rapyutasimulationplugins.readthedocs.io/en/latest/doxygen_generated/html/d2/dde/class_a_occupancy_map_generator.html) which generate png and yaml files for localization

james-yoo, 2023-06-11:
@YuminosukeSato,
I'd like to share my hands-on seminar resource on using rclUE and Nav 2 to navigate a custom environment. Although the material is written in Korean, I think you can understand it by looking at the pictures. You can see the Nav 2 integration on pages 22 through 33.

https://docs.google.com/presentation/d/1eZNXPmLtfYkiXLVbxpxOCY--ukxQFXVAd3Uaja0tKRQ/edit#slide=id.g20e9127e323_0_37

YuminosukeSato, 2023-06-11:
Thank you ! I'll try it.

YuminosukeSato, 2023-06-11:
I don't see rcIUE and RapyutaSimulationPlugins in the plugins, do I have to plug them in?
[image]
[image]

YuminosukeSato, 2023-06-11:
How to add rcIUE and RapyutaSimulationPlugins to plugins in unreal engine5?

YuminosukeSato, 2023-06-11:
I got error when I add RapyutaSimulationPlugins and rclUE as plugin.
I add RapyutaSimulationPlugins and rclUE in UnrealEngine/Engine/Plugin Folder and I do ./Setup.sh && ./GenerateProjectFiles.sh && make.
Then, I got error.
[error.txt](https://github.com/rapyuta-robotics/rclUE/files/11714059/error.txt)
How do I solve this error?

YuminosukeSato, 2023-06-14:
I have three questions.
First, what directories and files in tutlebot3-UE should I move into the field I created myself to make turtolbot3 appear?
Second, what directories and files in tutlebot3-UE should I move into the field I created so that I can plug in rcIUE and RapyutaSimulationPlugins?
Thirdly, will the level blueprint be the same as in tutlebot3-UE and will navigation in Ros2 work?

YuminosukeSato, 2023-06-17:
OccupancyGridMap 만들기
I am unable to create an OccupancyGridMap, I get an error when I place the OccupancyGridMap and press the play button.
Below is the error.
```
LoginId:d69c7d530464480cb426a9ddc047c2a3-000003e8
EpicAccountId:

Caught signal 11 Segmentation fault

libUnrealEditor-RapyutaSimulationPlugins.so!_ZN22AOccupancyMapGenerator9BeginPlayEv(+0x20)
libUnrealEditor-Engine.so!AActor::DispatchBeginPlay(bool) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/Actor.cpp:4053]
libUnrealEditor-Engine.so!AWorldSettings::NotifyBeginPlay() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/WorldSettings.cpp:291]
libUnrealEditor-Engine.so!AGameMode::HandleMatchHasStarted() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/GameMode.cpp:221]
libUnrealEditor-Engine.so!AGameMode::SetMatchState(FName) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/GameMode.cpp:339]
libUnrealEditor-Engine.so!AGameMode::StartPlay() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/GameMode.cpp:144]
libUnrealEditor-Engine.so!UWorld::BeginPlay() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/World.cpp:4881]
libUnrealEditor-Engine.so!UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Engine/Private/GameInstance.cpp:545]
libUnrealEditor-UnrealEd.so!UEditorEngine::CreateInnerProcessPIEGameInstance(FRequestPlaySessionParams&, FGameInstancePIEParameters const&, int) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:3210]
libUnrealEditor-UnrealEd.so!UEditorEngine::OnLoginPIEComplete_Deferred(int, bool, FString, FPieLoginStruct) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1672]
libUnrealEditor-UnrealEd.so!UEditorEngine::CreateNewPlayInEditorInstance(FRequestPlaySessionParams&, bool, EPlayNetMode) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1932]
libUnrealEditor-UnrealEd.so!UEditorEngine::StartPlayInEditorSession(FRequestPlaySessionParams&) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:2937]
libUnrealEditor-UnrealEd.so!UEditorEngine::StartQueuedPlaySessionRequestImpl() [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1237]
libUnrealEditor-UnrealEd.so!UEditorEngine::StartQueuedPlaySessionRequest() [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/PlayLevel.cpp:1149]
libUnrealEditor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/EditorEngine.cpp:1776]
libUnrealEditor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/home/iplslam/UnrealEngine/Engine/Source/./Editor/UnrealEd/Private/UnrealEdEngine.cpp:514]
UnrealEditor!FEngineLoop::Tick() [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Launch/Private/LaunchEngineLoop.cpp:5363]
UnrealEditor!GuardedMain(char16_t const*) [/home/iplslam/UnrealEngine/Engine/Source/./Runtime/Launch/Private/Launch.cpp:204]
libUnrealEditor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/iplslam/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:269]
libc.so.6!__libc_start_main(+0xf2)
UnrealEditor!_start()
```

yuokamoto, 2023-06-17:
Did you set Map?
https://rapyutasimulationplugins.readthedocs.io/en/latest/doxygen_generated/html/d2/dde/class_a_occupancy_map_generator.html#a6052b2bc9de6a01e06f37c2c565edb17

YuminosukeSato, 2023-06-17:
I didn't set Map. I try to set Map.But I don't get map image.
[Screencast from 2023年06月17日 17時01分12秒.webm](https://github.com/rapyuta-robotics/rclUE/assets/69001166/8a109119-c54e-4650-bcf0-3fbc98305c84)
[image]

yuokamoto, 2023-06-17:
Please check
- MaxVrticalHeight: https://rapyutasimulationplugins.readthedocs.io/en/latest/doxygen_generated/html/d2/dde/class_a_occupancy_map_generator.html#a4984d0bdf4e3c5347a20f3dbf1c03522
- Collision/Trace setting of wall and obstacle and bounding box of your Map. It uses tracing to generate occupancy map and that size is decide based on Map bounding box. https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/blob/8ff56736c0026f53f828075ece9af456e40e8f19/Source/RapyutaSimulationPlugins/Private/Tools/OccupancyMapGenerator.cpp#LL47C21-L47C38

YuminosukeSato, 2023-06-18:
I've created a floor tile with 3 tiles, but without one, it's normal map

YuminosukeSato, 2023-06-19:
The connection is made and the map is generated, but the waffle moves in the opposite direction of navigation.
[Screencast from 2023年06月19日 14時31分10秒.webm](https://github.com/rapyuta-robotics/rclUE/assets/69001166/daa66d78-101f-4fa9-8954-58532ee09c24)

YuminosukeSato, 2023-06-19:
I have one more question: we have attached a Yolo camera, but the camera does not work with waffle. How can I get it to work with turtolbot3?[Screencast from 2023年06月19日 15時25分49秒.webm](https://github.com/rapyuta-robotics/rclUE/assets/69001166/25c430a8-f319-4f44-a8c9-0fac936152fb)

yuokamoto, 2023-06-19:
Cool
- Did you check waffle move properly with cmd_vel?
- It is better to start with Turtlbot3WaffleVehicle, i.e. kinematic model since physics requires many parameter tunings
- `but the camera does not work with waffle`,  do you mean came do not move with tb3? did you check Mobility of camera?

YuminosukeSato, 2023-06-21:
Did you check waffle move properly with cmd_vel?　-> No. I don't know way to check waffle move properly with cmd_vel.
It is better to start with Turtlbot3WaffleVehicle, i.e. kinematic model since physics requires many parameter tunings->ok.I try it.
 but the camera does not work with waffle, do you mean came do not move with tb3?  -> Yes
did you check Mobility of camera?-> How can I check the mobility of my camera?

YuminosukeSato, 2023-06-21:
When I change to WAFFLE, objects do not land on the floor and fall straight down, how do I solve this?
[Screencast from 2023年06月21日 14時57分49秒.webm](https://github.com/rapyuta-robotics/rclUE/assets/69001166/62319b52-bb65-4c10-97ee-1ed5bced303f)

YuminosukeSato, 2023-06-21:
Only the Turtlbot3WaffleVehicle does not land on the floor and falls. How can I prevent it from falling during execution?
[Screencast from 2023年06月21日 14時57分49秒.webm](https://github.com/rapyuta-robotics/rclUE/assets/69001166/0511e31f-d1b9-4f34-9fb4-bfaa464317df)

james-yoo, 2023-06-21:
@YuminosukeSato 
You did not generate collision to static mesh(in this case floor).

YuminosukeSato, 2023-06-21:
The Turtlbot3BurgurVehicle lands firmly on the floor, but the Turtlbot3WaffleVehicle does not.

yuokamoto, 2023-06-21:
Please check collision setting of floor and obstacles
- Please check turltbot3-UE  map's floor collision setting and try to set simillar setting to your static meshes. ref https://docs.unrealengine.com/4.27/ja/InteractiveExperiences/Physics/Collision/Reference/

- You can visualize collision at runtime by
https://docs.unrealengine.com/4.27/ja/InteractiveExperiences/Physics/Collision/HowTo/PreviewCollision/

- about mobility
https://docs.unrealengine.com/4.27/ja/Basics/Actors/Mobility/

- you can check robot movement by using `ros2 run teleop_twist_keyboard teleop_twist_keyboard`

https://github.com/rapyuta-robotics/rclUE/assets/16463135/5a2d6126-3d5e-41eb-9e51-1f29d155b364

YuminosukeSato, 2023-06-24:
Thank you. I did it in my simulation. 
I have a question.
I would like to place multiple camera actors in UE5 and bring the data to OpenCV and SLAM together with Depth.Is it possible? Do you have a recommended GIThub repository?

yuokamoto, 2023-06-24:
- You can attach multiple cameras to robot and there is `ARRROS2CameraActor` which you can place as a stand alone actor. You need to change topic name and etc.
- There is a `URR3DLidarComponent` which publishes `PointCloud2`. But we haven't tested 3D lidar component much compared with 2DLidar
- Adding multiple camera may slow down simulation. There maybe more room to optimize process.

YuminosukeSato, 2023-06-25:
How can I use URR3DLidarComponent?

yuokamoto, 2023-06-25:
you can add 3D Lidar same way as camera

YuminosukeSato, 2023-06-26:
Thank you. 
Adding multiple cameras may slow down the simulation, so I would like to get the camera footage in unreal engine 5 and display it in python, but is it possible to get it via UDP communication or something?

yuokamoto, 2023-06-26:
not part of rclUE. you can check UE5 feature such as pixelstreaming and etc

YuminosukeSato, 2023-06-26:
ok.Thank you.

closed 2023-06-30 by yuokamoto