rapyuta-robotics/rclUE #98: Steps on working with ROS2 Humble, Ubuntu 20.04 and UE 4.27

ShadhyGit, 2023-06-27:
Hi,

I've managed to run the turtlebot example using ROS2 foxy and UE 4.27 but I need to use rclUE with ROS2 Humble and UE4.27.

How would I go about doing that? Thank you

yuokamoto, 2023-06-27:
Thanks for your question.
1. If you want to use msgs which is not changed between humble and foxy, I think you can keep using rclUE for foxy with humble since ROS 2 humble and foxy can be communicate. https://discourse.ros.org/t/ros-cross-distribution-communication/27335 
2. you can refer defference between `devel` and `UE5_devel_humble` since devel is for foxy and UE5_devel_humble is for humble. Theoritically it should work by applying same changes.

ShadhyGit, 2023-06-28:
Thank you for your answer. I'm trying to do what you described in step 2.

First I used the [UE_tools repository (humble branch)](https://github.com/rapyuta-robotics/UE_tools/tree/humble) to generate the ROS2 Humble files.
After this I updated the rclUE.Build.cs to include the humble files.

I am testing this in the turtlebot3-UE project and am not able to build the project with those changes.

Are you able to guide me towards the right steps to change from Foxy to Humble (for UE 4.27)?

yuokamoto, 2023-06-28:
- UE_tools devel branch should be able to use for humble by giving --rosdistro arg but may works only for UE5
- what is the build error you get?

ShadhyGit, 2023-06-28:
I used the devel branch and generated the ROS files. Here is the make_editor.sh output:
[make_editor_output.txt](https://github.com/rapyuta-robotics/rclUE/files/11892953/make_editor_output.txt)

After that I replaced the rclcutilities.h's contents with the one from the `UE5_devel_humble` branch. This is the output after that:

[make_editor_utilmod_output.txt](https://github.com/rapyuta-robotics/rclUE/files/11892980/make_editor_utilmod_output.txt)

yuokamoto, 2023-06-28:
hmm, I'm not sure where "unknown type name `turtlebot3_UE_Plugins_rclUE_Source_rclUE_Public_rclcUtilities_h_199_PROLOG`" comes from...

it may not very simple to achieve this. Are there any reason why you don't use option 1 in my answer?

ShadhyGit, 2023-06-30:
Hi, I was able to make it work by simply updating rclUtilities.h to match the one in the UE4 branch. 
I was working with the rclUE submodule in the turtlebot example which seems like it wasn't up to date.

To answer your question, I would have definitely used that option if I wasn't able to make it work.
Thanks for your help!

closed 2023-06-30 by ShadhyGit