rapyuta-robotics/rclUE #99: Custom Ros Msg generated not showing when ros2 topic list

yhg8423, 2023-06-28:
Hi, thank you for maintaining this repository for using ros2 in UE5.

Our team has used UE5.1 and ros2 foxy for our project.

In the project, we should use a custom ros msg, so we used `UE_tools` to generate the custom msg. 

It worked well, so we used the custom msg in the UE project, and we published the topic based on the custom msg using publisher blueprint.

However, it did not show the topic published by the blueprint when we used a ros2 command `ros2 topic list` in a terminal. 

Otherwise, when we published the topic with ros2 basic msg (e.g. twist), it showed the topic on a terminal using a ros2 command `ros2 topic list` . 

Furthermore, it also showed the publisher node when we used a ros2 command `ros2 node list` in a terminal.

How can I solve this issue?

yuokamoto, 2023-06-28:
Thanks for you question.
I'm not 100% sure but 
- Please double check that you execute `cd turtlebot3-UE && source fastdds_setup.sh` in your terminal which do `ros2 topic echo` 
- Please double check you `source colcon_ws/install/setup.bash` where you put your custom msg and colcon build.
- it woth testing by adding subscriber inside UE to check communication issue happen only between UE and non-UE.

yhg8423, 2023-06-29:
Thank you for your answer. 

We found that it does not work when using `Add Publisher` in Blueprint with Output Log `LogTemp: Error: [ROS2Publisher.cpp@76, virtual void UROS2Publisher::IntializeTopicComponent()] Failed Status on line 76 (function InitializeTopicComponent): 1. Continuning.` and `LogTemp: Error: [ROS2Publisher.cpp@138, void UROS2Publisher::Publish()] Failed status on line 138 (function Publish): 300. Continuing`.

Is there any way to address the errors?

yuokamoto, 2023-06-29:
line 76 is here
https://github.com/rapyuta-robotics/rclUE/blob/5d2767d0a1ea408e75b89ff6748fa401d2dd8dd2/Source/rclUE/Private/ROS2Publisher.cpp#L76

the error comes from rcl function, generally speaking, it is more difficult to debug.

Could you share the screenshot of blueprint if possible?

yuokamoto, 2023-06-29:
FYI:
I recently created PR to update ue_msgs.
https://github.com/rapyuta-robotics/rclUE/pull/100/files

simillar file changes should happen with your custom msg generation.

yhg8423, 2023-06-29:
Thank you for your answers!

We solved this problem with execute `source colcon_ws/install/setup.bash` in terminal before launching the UE5 execution file, and we executed the UE5 using the terminal which already execute the command `source colcon_ws/install/setup.bash`. 

If anyone wants to use custom msgs in UE, I recommend to sourcing `install/setup.bash` in ros workspace in the terminal and executing UE5 using the terminal which already `install/setup.bash` in the ros workspace!

closed 2023-06-29 by yhg8423

mhl787156, 2024-07-25:
Hi @yhg8423 I'm also interested in using custom ROS msgs within the UE5 environment, but I'm a it confused at how to use UE_TOOLS. In particular, if I have a ros2 pkg `my_custom_msgs` in a ros workspace `my_ros2_ws`, how do I use `python3 docker_build_install_codegen.py ...` to generate it over. 

I'm getting a bit confused between that tool and the `build_install_codegen.py` command with instructions here (https://github.com/rapyuta-robotics/UE_tools?tab=readme-ov-file#add-custom-lib-to-your-plugin-1)!  

It seems you've managed to parse the instructions, so I was hoping you might be able to point me in the right direction! Many Thanks!

yhg8423, 2024-10-09:
Hi, @mhl787156 . I am sorry to late comment. 
Did you solve the problem?

It's been a while, so my memory isn't perfect, but in my case I didn't use docker.
I only used below commands.

```
python3 build_install_codegen.py --type pkgs --install --rosdistro foxy --skip_pull --config custom_config.yaml 
python3 build_install_codegen.py --type pkgs --codegen --rosdistro foxy --skip_pull --config custom_config.yaml 

```

I apologize for not being of much help. Thanks!