rapyuta-robotics/rclUE #133: Usage without a fixed framerate

alassagne-sl, 2024-02-02:
Hello,
I'm using this Plugin with ROS Humble / Unreal 5.3.2 and it works very nicely. Thank you for that!

I wonder if it's possible to run a project without the FixedFramerate constraint. In certain cases I'd like to be live and not in simulation time, dropping frames instead of slowing down. Also the synchronization mecanisms is pretty hard on the performance, the same scene in another project is much faster.

Is there a way to disable this behaviour?

yuokamoto, 2024-02-02:
Hello, thanks for your question.

I think you can run project without fixed frame rate by changing setting around here
https://github.com/rapyuta-robotics/turtlebot3-UE/blob/6f118f8a08f5b5251e458b014d9a8a03c35736ab/Config/DefaultEngineBase.ini#L19

```
bUseFixedFrameRate=True
FixedFrameRate=${FIXED_FRAME_RATE}
TargetRTF=${TARGET_RTF}
CustomTimeStepClassName=/Script/RapyutaSimulationPlugins.RRLimitRTFFixedSizeCustomTimeStep
```

braoutch, 2024-02-05:
Hi, thank you for your response. Simply changing this setting did not work for me. I don't have much time this week but I'll investigate more and report my conclusions in this thread.

yuokamoto, 2024-02-06:
Could you please check project setting as well?
[image]

Bvallon-sl, 2024-02-08:
Hi, 

We want to not use the CutomTimeStep, but when we remove it from the project settings and the .ini files, it's still used.
Do you have any idea why? Maybe, it's set in the code but I did not find it.

Thanks again for your help and your project, it's amazing.

Best.

Bvallon-sl, 2024-02-08:
Hi,

Thanks for the help but I no longer use the run_editor script because It overrides the project settings at each start.
In the config file, I even set` CustomTimeStepClassName=None` but It looks like it's still used, that's why I was wondering if it was enabled somewhere else.

yuokamoto, 2024-02-09:
Did you try change CUstomTimeStepClassName from `Project Setting` in the editor before play?  
https://docs.unrealengine.com/4.27/en-US/Basics/Projects/ProjectSettings/

SLJLMacarit, 2024-06-18:
> Did you try change CUstomTimeStepClassName from Project Setting in the editor before play?

This did it for me. What I did, to sum up:
- change this timestep class to none
- make sure that ARRROS2GameMode::SetFixedTimeStep is not called in code or blueprint (I even commented out what's inside to be extra sure, but I don't think you want to do that :sweat_smile: )
- comment out the line in `run_editor.sh` that was resetting the `DefaultEngine.ini`
- interestingly, I also had to comment out most of `URRLimitRTFFixedSizeCustomTimeStep`'s constructor, which I suppose is called somewhere I didn't find. It's not a problem for the moment, I'm a bit worried about implications I don't see, but it seems to work for now.

Thanks a lot, @yuokamoto, @james-yoo !

closed 2024-06-22 by yuokamoto