rapyuta-robotics/rclUE #134: QoS profiles available in the plugin

dbarrita, 2024-02-15:
Hi, I'm using the plugin and I need fast communication in my publishers and subscribers. I have seen that there are different QoS profiles available in the plugin that appear in the [ROS2 documentation](https://docs.ros.org/en/humble/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-profiles): Default, Services, Sensor Data, Parameters and System Default. However, there are other options available such as: ClockPub, StaticBroadcaster, DynamicBroadcaster, UnknowQoS which are not defined. Could you specify what is the behavior of these profiles or point me to the documentation that describes these profiles? Thanks.

yuokamoto, 2024-02-15:
Hi, thanks for your question.
We haven't add those in documentation but you can find definition here in source code
https://rclue.readthedocs.io/en/devel/doxygen_generated/html/d8/d54/rclc_utilities_8h_source.html

which follows  rmw definition here 
http://docs.ros.org/en/humble/p/rmw/generated/file_include_rmw_qos_profiles.h.html#file-include-rmw-qos-profiles-h

dbarrita, 2024-02-16:
Hi, thanks for the quick reply! Now I have enough information to choose the best option for my problem.

dbarrita, 2024-02-27:
Hi again, I was checking the different QoS profiles in the source code [(link)](https://rclue.readthedocs.io/en/devel/doxygen_generated/html/d8/d54/rclc_utilities_8h_source.html) and I see that in this plugin there is no option for a profile with reliability "best effort". However, this policy is set in the ROS2 profiles [(link)](https://github.com/ros2/rmw/blob/humble/rmw/include/rmw/qos_profiles.h) in the sensor_data profile. Is there a reason for not having a best effort option? Thanks in advance!

yuokamoto, 2024-02-28:
Honestly says, I don't remember...
Maybe just a bug, we can create PR to fix to use best effort

dbarrita, 2024-02-28:
Thanks for the answer, it would be interesting if you could create the PR to have the sensor data with best effort option, because is the only profile that use best effort for this policy.

yuokamoto, 2024-02-28:
https://github.com/rapyuta-robotics/rclUE/pull/135

closed 2024-02-29 by dbarrita