rapyuta-robotics/rclUE #137: Android Support

adarshjamuaar, 2024-03-15:
Is there any way to include this plugin with android build? Do I need to recompile ROS libraries?
Can someone please give me a step-wise process to make it compatible? 
Thanks!

yuokamoto, 2024-03-15:
Hello,

I think yes, if you can build ros for android and put into [ThirdParty](https://github.com/rapyuta-robotics/rclUE/tree/devel/ThirdParty/ros), it should work but I haven't tried yet.

adarshjamuaar, 2024-03-15:
@yuokamoto thanks for your reply! is there any guide to build ros for android that I can follow?

yuokamoto, 2024-03-15:
I don't have experience on android development, but I found this https://github.com/ros2-java/ros2_java by google

adarshjamuaar, 2024-03-16:
ah it uses fast-rtps and not fast-dds as this plugin does I suppose..

adarshjamuaar, 2024-04-01:
hi any other help?

yuokamoto, 2024-04-16:
Sorry, I don't have Android device and can't test it quickly...