rapyuta-robotics/rclUE #138: Make RosFluidPressure: Missing Connector for pressure and variance

LukasD97, 2024-04-04:
[image]
The Connector to set the Pressure is missing in the Blueprint Node.
Is there an option i missed out or is this an bug?
Thanks!

LukasD97, 2024-04-04:
Solved by modifiing UPROPERTY(EditAnywhere) -> UPROPERTY(EditAnywhere, BlueprintReadWrite)
in rclUE/Source/rclUE/Public/Msgs/ROS2FluidPressure.h

yuokamoto, 2024-04-04:
Hello, it is same issue as this
https://github.com/rapyuta-robotics/RapyutaSimulationPlugins/issues/271#issuecomment-1997416677

 BP does not support double

LukasD97, 2024-04-04:
Hi, thanks. 
For me double works after the change.
[image]

yuokamoto, 2024-04-04:
Oh, interesting. Thanks for letting us know.
I will check and make all doubles accesible from BP if possible

LukasD97, 2024-04-04:
No Problem :)
This would be nice

yuokamoto, 2024-04-06:
I created PR
https://github.com/rapyuta-robotics/rclUE/pull/139

closed 2024-04-12 by yuokamoto