rapyuta-robotics/rclUE #140: Is there an possibility to have an higher publish rate than the fps

LukasD97, 2024-04-10:
Currently it seems like the maximum Publish rate of an Node is bound to the frame rate. 
Is there an option to allow higher pub rates?

yuokamoto, 2024-04-12:
Currently there is no option to pub since it does not make sense update information higher than simulation update.
Are there any use case to publish higher than frame rate?

james-yoo, 2024-04-12:
Generally IMU sensor publish acceleration and gyro in high frequency(200hz, 400hz).

yuokamoto, 2024-04-16:
hmm, I see
but since it is higher than frame rate(higher than simulation world state update), it will result in publishing same contents in same frame.
By customizing publisher, it can be possible to publish multiple times in one frame but not sure it make sense or not

yuokamoto, 2024-04-17:
- which plugin?
- you can publish manually please refer. you can publish multiple times in one tick
https://github.com/rapyuta-robotics/turtlebot3-UE/blob/6f118f8a08f5b5251e458b014d9a8a03c35736ab/Source/turtlebot3/ROS2Examples/ROS2PublisherNode.cpp#L35

yuokamoto, 2024-04-18:
bp
https://rclue.readthedocs.io/en/devel/examples/publisher_example.html#bp-publisher

yuokamoto, 2024-09-12:
What do you mean only 60 possible? Some param's maximum is 60 or simulation run with 60 hz because of CPU/GPU limitation?

LukasD97, 2024-09-12:
60 hz publish rate, cpu and gpu arent an limitation. 60 fps is ok for me, i only want to publish at an higher rate

yuokamoto, 2024-09-12:
Where do you set 60hz?
you can publish as many times as you want by call Publish in your tick

LukasD97, 2024-09-12:
I've set 240 hz in the publishers. But they're only publishing with 60 hz

yuokamoto, 2024-09-12:
maximum Publish rate of an Node is bound to the frame rate as you mentioned.

You need to create non loop publisher and manually call Publish() from your bp to meet your requirements.
https://rclue.readthedocs.io/en/devel/examples/publisher_example.html#id1

Zartris, 2025-02-15:
Sorry to join the conversation out of nowhere. Where do we see this limit today exactly? 
Is it the publisher class that is bound by the framerate, because i cannot seem to find something pointing to this. 
Is it the ROS2_LOOP_PUBLISHER_WITH_QOS that only runs doing update calls?

LukasD97, 2025-02-15:
The Non loop publisher has no limit. The Jsbsim plugin is only updating its data at 60 hz so it's the limitng factor.

Zartris, 2025-02-16:
Ah so another plugin is the bottleneck, got it! I also thought it was weird as I did not see any problem in your code, both in the sensor and publisher side.

I guess it is time to just install it and get a feel for it myself 👍