rapyuta-robotics/rclUE #150: Missing documentation of libyaml.so and libspdlog.so.1

PhDittmann, 2024-08-08:
Due to pre-compiled libraries, `ThirdParty/ros/lib/librcl.so` dynamically links `libyaml.so` and `libspdlog.so.1`, which needs to be provided by/installed on the host system. If not, Unreal fails to load the plugin or package the project without further details.

On some operating systems, even with libyaml and libspdlog installed, the version appendix may does not exist. You can try creating them using:
```shell
cd /lib64
ln -s </path/to/libyaml.so.X.Y.Z> libyaml.so
ln -s </path/to/libspdlog.so.X.Y.Z> libspdlog.so.1
```

yuokamoto, 2024-10-02:
sorry for late response.
I;ll check

yuokamoto, 2024-10-08:
Aren't libyaml.so and libspdlog.so.q installed when you install ROS 2?

PhDittmann, 2024-10-23:
Maybe, but it does not work if ROS 2 runs on a different machine on i.e. a docker container.

yuokamoto, 2024-10-29:
Updated README as you suggested.
Thanks for your help.

https://github.com/rapyuta-robotics/rclUE

closed 2024-10-29 by yuokamoto