rapyuta-robotics/rclUE #154: Action example

ArjanVroegop, 2024-10-09:
Hi,

Great work on rclUE, I'm really impressed by how well ROS2 is integrated in Unreal Engine.

I have a question about the ROS2ActionExamples. If I open this level, everything works fine within UE. However, I have some issues talking to an action server from ROS2.

To reproduce:
- Open the ROS2ActionExamples level
- Delete everything except BP_ROS2ActionServerNode
- Play the level

 From a terminal, with ROS2 sourced,  `ros2 action list` shows `/bp/fibonacci`. So the action server seems to be there.
However, with the command `ros2 action send_goal /bp/fibonacci example_interfaces/action/Fibonacci "{}"` I only get this output: `Waiting for an action server to become available...`
Could this be a bug in the ActionServer or am I doing something wrong?

yuokamoto, 2024-10-10:
Let me check.

yuokamoto, 2024-10-12:
I can reproduce issue. I will keep debugging but currently no clue...

ArjanVroegop, 2024-10-16:
Hi yuokamoto, thanks for looking into this. Is there anything I can do to help you to find the problem?

yuokamoto, 2024-10-17:
Thank you. I spent a time to investigate but can't find clue yet. Since it is ROS 2 communication issue, it may need to check deep inside.

titou-B, 2025-06-12:
Hi ! And thanks for the great work !

I have the same problem and I noticed something strange by chance: if I launch an action server from a terminal in parallel with the one from rclUE in UE, my rclUE action server receives the goal and I get the following log:
```
LogROS2Action: [ROS2ActionServer.cpp@72, virtual void UROS2ActionServer::ProcessReady(rcl_wait_set_t *)] 2. Action Server - Received goal request
LogROS2Action: [ROS2ActionServer.cpp@124, void UROS2ActionServer::SendGoalResponse()] 3. Action Server - Send goal response
LogTemp: Error: [ROS2ActionServer.cpp@132, void UROS2ActionServer::SendGoalResponse()] Failed status on line 132 (function SendGoalResponse): 1. Continuing.
LogRos2SandBox: [ROS2ActionServerNode.cpp@86, void AROS2ActionServerNode::GoalCallback(UROS2GenericAction *)] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][goal callback]
LogROS2Action: [ROS2ActionServer.cpp@88, virtual void UROS2ActionServer::ProcessReady(rcl_wait_set_t *)] 6. Action Server - Received result request
LogRos2SandBox: [ROS2ActionServerNode.cpp@114, void AROS2ActionServerNode::ResultCallback()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][result callback] Start fibonacci calculation
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 1
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 2
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 3
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 5
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 8
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogRos2SandBox: [ROS2ActionServerNode.cpp@43, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][update feedback] added 13
LogROS2Action: [ROS2ActionServer.cpp@137, void UROS2ActionServer::SendFeedback()] 7. Action Server - Publish feedback
LogROS2Action: [ROS2ActionServer.cpp@150, void UROS2ActionServer::SendResultResponse()] 9. Action Server - Send result response
LogTemp: Error: [ROS2ActionServer.cpp@158, void UROS2ActionServer::SendResultResponse()] Failed status on line 158 (function SendResultResponse): 1. Continuing.
LogRos2SandBox: [ROS2ActionServerNode.cpp@67, void AROS2ActionServerNode::Execute()] [ROS2ActionServerNode_UAID_A0CEC81436F66C7102_1266473832] [fibonacci_action][C++][send result] result is: 0, 1, 1, 2, 3, 5, 8, 13, 
``

Using Ubuntu 22.04, UE5.5 and rclUE is on UE5.5_devel_humble branch