rapyuta-robotics/rclUE #165: Mac Support

ei06125, 2025-01-31:
My apologies if this is a skill issue but I need to know if this library supports MacOS or not.
I know the README.md has a Main Support section but maybe there is a branch where support is being tested.
If not, is there any hint you can give me to make it work?

I have been looking into the code base and of course shared libraries might be an initial problem - converting all of them to MacOS type. But aside from that, anything I should look into? Cross-platform programming if using the standard library should be expected, yes?

Thank you

yuokamoto, 2025-01-31:
Hi, Thanks for having intersted in this repo.

Unfortunatelly rclUE support only Ubuntu and some voluteer make it work in Windows but I haven't tested.
ROS 2 itslef 's support of Mac is Tier3
https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027

As you wrote, theoritically build lib for mac and put inside thirdparty folder and link properly should make it work in Mac but not sure how much difficult is it honestly...

ei06125, 2025-01-31:
Might be too much.
Just one more question: what are the exact dependencies of rclUE? If I was to pursue this MacOS journey, what ros library was included? ROS1? ROS2? rclcpp? rcl?