rapyuta-robotics/rclUE #166: Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

1157457050, 2025-02-08:
[2025.02.08-08.29.42:824][ 78]LogROS2Node: [ROS2NodeComponent.cpp@109, void UROS2NodeComponent::Init()] [ROS2Node] start initializing..
[2025.02.08-08.29.42:824][ 78]LogROS2Node: [ROS2NodeComponent.cpp@114, void UROS2NodeComponent::Init()] [ROS2Node] rclc_node_init_default
Signal 11 caught.
Malloc Size=262146 LargeMemoryPoolOffset=262162 
CommonUnixCrashHandler: Signal=11
[2025.02.08-08.29.42:911][ 78]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x000000000000000f

[2025.02.08-08.29.42:911][ 78]LogCore: Fatal error!

0x0000733cf5d9d7fd libc.so.6!UnknownFunction(0x19d7fd)
0x0000733cf5c76d31 libc.so.6!UnknownFunction(0x76d30)
0x0000733cf5c8849a libc.so.6!UnknownFunction(0x88499)
0x0000733cf5c60856 libc.so.6!snprintf(+0x95)
0x0000733c892e53c1 librmw_fastrtps_shared_cpp.so!_ZN23rmw_fastrtps_shared_cpp18create_participantEPKcmPK22rmw_security_options_sbS1_PN14rmw_dds_common7ContextE(+0x2a0)
0x0000733c895c97c4 librmw_fastrtps_cpp.so!UnknownFunction(0x327c3)
0x0000733c895c95e8 librmw_fastrtps_cpp.so!UnknownFunction(0x325e7)
0x0000733c895e6ae6 librmw_fastrtps_cpp.so!rmw_create_node(+0x255)
0x0000733cf5e9169d librmw_implementation.so!rmw_create_node(+0x6c)
0x0000733cf7e1f461 librcl.so!rcl_node_init(+0xac0)
0x0000733cf7d0b0c6 librclc.so!rclc_node_init_with_options(+0x1a5)
0x0000733cf7d0adb2 librclc.so!rclc_node_init_default(+0x171)
0x000000000edd2315 CarlaUnreal!UROS2NodeComponent::Init() [/home/carlaue5/Sim/CarlaUE5_New/Unreal/CarlaUnreal/Plugins/rclUE-UE5_devel_humble/Source/rclUE/Private/ROS2NodeComponent.cpp:116]
0x0000000006c89067 CarlaUnreal!UFunction::Invoke(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/Class.cpp:7192]
0x0000000006ee05e9 CarlaUnreal!UObject::CallFunction(FFrame&, void*, UFunction*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1147]
0x0000000006ee9d69 CarlaUnreal!UObject::ProcessContextOpcode(FFrame&, void*, bool) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006ee109b CarlaUnreal!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006ee07f7 CarlaUnreal!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1040]
0x0000000006ee1991 CarlaUnreal!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*)::$_0::operator()() const [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1286]
0x0000000006ee18b2 CarlaUnreal!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1302]
0x0000000006ee109b CarlaUnreal!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006c89067 CarlaUnreal!UFunction::Invoke(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/Class.cpp:7192]
0x0000000006ee5311 CarlaUnreal!UObject::ProcessEvent(UFunction*, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2173]
0x000000000bff05bd CarlaUnreal!AActor::ProcessEvent(UFunction*, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/Actor.cpp:1160]
0x000000000c00fcc4 CarlaUnreal!AActor::BeginPlay() [/home/carlaue5/Sim/CarlaUE5_New/Unreal/CarlaUnreal/Intermediate/Build/Linux/CarlaUnreal/Inc/Engine/UHT/Actor.gen.cpp:7876]
0x000000000c00f41f CarlaUnreal!AActor::DispatchBeginPlay(bool) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/Actor.cpp:4282]
0x000000000c00eba0 CarlaUnreal!AActor::PostActorConstruction() [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/Actor.cpp:4070]
0x000000000c0018c0 CarlaUnreal!AActor::FinishSpawning(UE::Math::TTransform<double> const&, bool, FComponentInstanceDataCache const*, ESpawnActorScaleMethod) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/Actor.cpp:3969]
0x000000000c748e1a CarlaUnreal!UGameplayStatics::FinishSpawningActor(AActor*, UE::Math::TTransform<double> const&, ESpawnActorScaleMethod) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/GameplayStatics.cpp:893]
0x000000000c754e23 CarlaUnreal!UGameplayStatics::execFinishSpawningActor(UObject*, FFrame&, void*) [/home/carlaue5/Sim/CarlaUE5_New/Unreal/CarlaUnreal/Intermediate/Build/Linux/CarlaUnreal/Inc/Engine/UHT/GameplayStatics.gen.cpp:2707]
0x0000000006ee0073 CarlaUnreal!UObject::execCallMathFunction(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1079]
0x0000000006ee9310 CarlaUnreal!UObject::execLetObj(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006ee109b CarlaUnreal!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006ee07f7 CarlaUnreal!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1040]
0x0000000006ee1991 CarlaUnreal!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*)::$_0::operator()() const [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1286]
0x0000000006ee18b2 CarlaUnreal!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1302]
0x0000000006ee109b CarlaUnreal!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:482]
0x0000000006c89067 CarlaUnreal!UFunction::Invoke(UObject*, FFrame&, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/Class.cpp:7192]
0x0000000006ee5311 CarlaUnreal!UObject::ProcessEvent(UFunction*, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2173]
0x000000000bff05bd CarlaUnreal!AActor::ProcessEvent(UFunction*, void*) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/Actor.cpp:1160]
0x000000000767a419 CarlaUnreal!void TScriptDelegate<FNotThreadSafeDelegateMode>::ProcessDelegate<UObject>(void*) const [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/Runtime/Core/Public/UObject/ScriptDelegates.h:448]
0x000000000c48828d CarlaUnreal!FInputActionHandlerDynamicSignature_DelegateWrapper(TScriptDelegate<FNotThreadSafeDelegateMode> const&, FKey) [/home/carlaue5/Sim/CarlaUE5_New/Unreal/CarlaUnreal/Intermediate/Build/Linux/CarlaUnreal/Inc/Engine/UHT/InputComponent.gen.cpp:73]
0x000000000d41bf38 CarlaUnreal!FInputActionUnifiedDelegate::Execute(FKey) const [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/Runtime/Engine/Classes/Components/InputComponent.h:127]
0x000000000d41a9aa CarlaUnreal!UPlayerInput::EvaluateInputDelegates(TArray<UInputComponent*, TSizedDefaultAllocator<32> > const&, float, bool, TArray<TTuple<FKey, FKeyState*>, TSizedDefaultAllocator<32> > const&) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./Runtime/Engine/Private/UserInterface/PlayerInput.cpp:1553]
0x000000000ef232a3 CarlaUnreal!UEnhancedPlayerInput::EvaluateInputDelegates(TArray<UInputComponent*, TSizedDefaultAllocator<32> > const&, float, bool, TArray<TTuple<FKey, FKeyState*>, TSizedDefaultAllocator<32> > const&) [/home/carlaue5/Sim/UnrealEngine5_carla/Engine/Source/./../Plugins/EnhancedInput/Source/EnhancedInput/Private/EnhancedPlayerInput.cpp:776]
[... cut ...]
[2025.02.08-08.29.42:912][ 78]LogExit: Executing StaticShutdownAfterError
Malloc Size=131160 LargeMemoryPoolOffset=393352 
Malloc Size=131160 LargeMemoryPoolOffset=524536 
Malloc Size=136641 LargeMemoryPoolOffset=661201 
[2025.02.08-08.29.42:923][ 78]LogHAL: Error: FUnixPlatformProcess::CreateProc: File does not exist (/home/carlaue5/Sim/CarlaUE5_New/Build/Package/Carla-0.10.0-Linux-Development/Linux/Engine/Binaries/Linux/CrashReportClient)
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

yuokamoto, 2025-02-08:
Hello,
Are you using turtlebot3-UE?
Could you share the Ubuntu distribution of your environment?

1157457050, 2025-02-09:
> Hello, Are you using turtlebot3-UE? Could you share the Ubuntu distribution of your environment?

UE5.5.0 Ubuntu22.04

yuokamoto, 2025-02-09:
Could you try jammy_UE5.5 branch in turtlebot3-UE?